RobertCNelson/bb.org-overlays #4: Running as PWM

larivierec, 2015-08-29:
Hello everyone,

I recently updated my Kernel to 4.1.x and found that is a lot simpler to use the gpios this way and it seems to work great with the univ-all.dts

However, this dts does not allow me to take advantage of any pwm pins as almost every pin is setup as a digital gpio.

What would be the correct way to enable the 8 pwm pins?

Thanks in advance,

spmaniato, 2016-03-18:
I'm having the same problem under the 4.1.17 Kernel (I'm on Ubuntu 14.04.4 for ARMhf). Specifically, there are no `bone_pwm*` device tree overlays under `/lib/firmware` (or any overlays containing `pwm` for that matter). Any suggestions on how to proceed? Thank you :smile:

RobertCNelson, 2016-03-18:
@spmaniato capes are entering this repo on a test by test basis..

Which "bone_pwm_xyz" are you willing to test right now?

Regards,

spmaniato, 2016-03-18:
Oh, I see. I'd gladly test `am33xx_pwm` and`bone_pwm_P9_14`

So far I've been able to compile the latter based on this DTO generator: http://kilobaser.com/blog/2014-07-28-beaglebone-black-devicetreeoverlay-generator

RobertCNelson, 2016-03-18:
Okay i pushed it to:

https://github.com/beagleboard/bb.org-overlays

I'm not sure:

```
	fragment@1 {
		target-path="/";
		__overlay__ {
			pwm_test_P9_14 {
				compatible	= "pwm_test";
				pwms 		= <&ehrpwm1 0 500000 1>;
				pwm-names 	= "PWM_P9_14";
				pinctrl-names	= "default";
				pinctrl-0	= <&pwm_P9_14>;
				enabled		= <1>;
				duty		= <0>;
				status 		= "okay";
			};
		};
	};
```

Is actually going to work as pwm-test isn't mainline..

Regards,

RobertCNelson, 2016-03-18:
I "think" it would be best to model these after a pwm-led:

```
    fragment@1 {
        target-path="/";
        __overlay__ {
        pwm {
                compatible = "pwm-leds";

                pwm_P9_14 {
                    label  = "PWM_P9_14";
                    pwms        = <&ehrpwm1 0 500000 1>;
                    max-brightness = <255>;
                    linux,default-trigger = "default-on";
                };
            };
        };
    };
```

```
	fragment@4 {
		target = <&ehrpwm1>;
		__overlay__ {
			pinctrl-names   = "default";
			pinctrl-0   = <&pwm_P9_14>;
			status = "okay";
		};
	};
```

and then the pinmux would h

spmaniato, 2016-03-18:
Thanks! I'll report back.

spmaniato, 2016-03-18:
Quick update: 
* The `am33xx_pwm-00A0.dts` overlay compiled as is.
* The `bone_pwm_P9_14-00A0.dts` is giving me syntax errors:
```bash
/lib/firmware$ sudo dtc -O dtb -o bone_pwm_P9_14-00A0.dtbo -b 0 -@ bone_pwm_P9_14-00A0.dts
Error: bone_pwm_P9_14-00A0.dts:12.2-9 syntax error
FATAL ERROR: Unable to parse input tree
```
I tried the (unfamiliar to me) syntax: `/include/ "dt-bindings/board/am335x-bbw-bbb-base.h"` but then it couldn't find the header file. Not sure this syntax is even applicable here.
```
FATAL ERROR: Couldn't open "dt-bindings/board/am335x-bbw-bbb-base.h": No such file or directory
```
I eventually got it to compile by getting rid of the header files and replacing `BONE_P9_14` with `0x048`

I'm gonna try controlling a stepper motor next.

PS. In case it's relevant, here's my `dtc` version:
```
$ dtc --version
Version: DTC 1.4.1-g1e75ebc9
```

RobertCNelson, 2016-03-18:
it's setup to build inside the repo:

https://github.com/beagleboard/bb.org-overlays/blob/master/readme.md

Regards

spmaniato, 2016-03-18:
> it's setup to build inside the repo

Classic case of [RTFM](https://en.wikipedia.org/wiki/RTFM). Thanks!

After installing the device tree blobs and rebooting, I can load the two overlays:
```
$ cat $SLOTS
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
13: P-O-L-   0 Override Board Name,00A0,Override Manuf,bone_pwm_P9_14
14: P-O-L-   1 Override Board Name,00A0,Override Manuf,am33xx_pwm
```
Also, `pwm_test_P9_14` appeared under `/sys/devices/platform`. But:
```
ls -lh /sys/devices/platform/pwm_test_P9_14
total 0
-rw-r--r-- 1 root root 4.0K Mar 18 19:02 driver_override
-r--r--r-- 1 root root 4.0K Mar 18 19:02 modalias
lrwxrwxrwx 1 root root    0 Mar 18 19:02 of_node -> ../../../firmware/devicetree/base/pwm_test_P9_14
drwxr-xr-x 2 root root    0 Mar 18 19:02 power
lrwxrwxrwx 1 root root    0 Mar 18 19:02 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4.0K Mar 18 19:00 uevent
```
I was expecting to see things like `duty`, `period`, and `run` there. Well, actually I was expecting them under `/sys/devices/platform/ocp`, but there's no  `pwm_test_P9_14` there.

Please let me know if I'm going about it the wrong way. I'm basically trying to pattern-match looking at overlay / PWM instructions for older Kernels :weary:

In the meantime, I'll revisit your suggestion above (to model these after a pwm-led).

RobertCNelson, 2016-03-18:
Yeah, that's the issue, "pwm-test" isn't a valid option past 3.8.x, it was something special..

On teh "bacon" cape which used 3 pwm pins, i had to setup it up like:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BONE-BACONE-00A0.dts#L138-L145

Then you had to manually export the pwm like:

https://github.com/beagleboard/bb.org-overlays/blob/master/examples/BB-BONE-BACONE/example.sh

i think modeling this as  leds-pwm, would work best, but we need to bring out the (period=cycle) and duty values as a dt option..

Regards,

spmaniato, 2016-04-12:
The latest pull request, #6, adds PWM functionality to the new kernel, right? 😀

RobertCNelson, 2016-04-12:
I believe so, haven't had to test it yet. @Pillar1989 been connecting the userspace library/appliction mraa/upm to these pwm/usart/i2c overlays.