RobertCNelson/bb.org-overlays #9: Load a dto at startup

TigerClaw33, 2016-08-21:
Hi,
I try to load the dto for the I2C-1-Bus and a second dto for my extern ADC.
`cape_enable=bone_capemgr.enable_partno=BB-I2C1 `
works.

`cape_enable=bone_capemgr.enable_partno=BB-I2C1 , BB-ADC-GPIO`

didn't work. Only BB.I2C1 is loaded:

```
root@beaglebone:/# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-I2C1
```
And I don't want to execute the following command after every startup:

```
root@beaglebone:/# sudo sh -c "echo 'BB-ADC-GPIO' > /sys/devices/platform/bone_capemgr/slots"
root@beaglebone:/# cat /sys/devices/platform/bone_capemgr/slots                  0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-I2C1
 6: P-O-L-   1 Override Board Name,00A0,Override Manuf,BB-ADC-GPIO
```

Why the command in the /boot/uEnv.txt is only working for BB-I2C1 but not for the second dto?

Thanks.

RobertCNelson, 2016-08-21:
Well, drop the spaces...

```
cape_enable=bone_capemgr.enable_partno=BB-I2C1,BB-ADC-GPIO
```

TigerClaw33, 2016-08-21:
This was a copy and space mistake. They are drop

RobertCNelson, 2016-08-22:
Make sure the initrd.img has your addon "BB-ADC_GPIO" *.dtbo:

```
sudo update-initramfs -uk `uname -r`
```

Regards,

TigerClaw33, 2016-08-22:
Thanks, I tried this a few weeks ago but maybe forgot to set "-t" to override. 

`sudo update-initramfs -uk `uname -r` -t`

works!