RobertCNelson/bb.org-overlays #12: Cannot load 'BB-UART1' - "Failed Verification"

altr0n, 2017-03-28:
I get the following error when I try to follow the readme:

root@beaglebone:/home/debian#  echo 'BB-UART1' > /sys/devices/platform/bone_capemgr/slots
bash: echo: write error: File exists

Kernel version:

Linux beaglebone 4.4.54-ti-r93 #1 SMP Fri Mar 17 13:08:22 UTC 2017 armv7l GNU/Linux

Dmesg Output:

[  522.701595] bone_capemgr bone_capemgr: part_number 'BB-UART1', version 'N/A'
[  522.701680] bone_capemgr bone_capemgr: slot #6: override
[  522.701723] bone_capemgr bone_capemgr: Using override eeprom data at slot 6
[  522.701770] bone_capemgr bone_capemgr: slot #6: 'Override Board Name,00A0,Override Manuf,BB-UART1'
[  522.712138] bone_capemgr bone_capemgr: slot #6: BB-UART1 conflict P9.24 (#4:cape-universaln)
[  522.720969] bone_capemgr bone_capemgr: slot #6: Failed verification

I've not been able to find any resources online to help me resolve this problem and did not know where else to ask. Any thoughts are greatly appreciated as I've been stuck on this for a few days now.

RobertCNelson, 2017-03-28:
```
[ 522.712138] bone_capemgr bone_capemgr: slot #6: BB-UART1 conflict P9.24 (#4:cape-universaln)
[ 522.720969] bone_capemgr bone_capemgr: slot #6: Failed verification
```
cape-universaln is already loaded to bb-uart1 can't load.. To disable this, open /boot/uEnv.txt and remove "cape_universal=enable" and reboot.

Regards,

closed 2017-03-28 by RobertCNelson

altr0n, 2017-03-28:
Thank you for clearing this up - I'm sorry for wasting your time!