RobertCNelson/bb.org-overlays #13: Debian packaging doesn't install tools/dtbo

malc0, 2017-04-20:
Not having the dtbo script as an initramfs hook means that *.dtbo doesn't get included in the initrd's /lib/firmware by update-initramfs, and bone_capemgr.enable_partno= and friends don't work.

RobertCNelson, 2017-04-21:
@malc0 for historical reasons it's in the bb-customizations package.

https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/jessie/debian/dtbo

now that we have a separate overlay package i should merge that script into this repo.

Regards,

malc0, 2017-04-21:
Aha, thanks for the pointer, and apologies for not filing this against the repos repo.  I'd done a stretch netinstall and ended up with neither bb.org-overlays nor bb-customizations.  Perhaps, if merging the script has any complications, the situation I'd created could be avoided by making bb.org-overlays depend on bb-customizations?

RobertCNelson, 2017-04-21:
i came up with this:

https://github.com/rcn-ee/repos/blob/master/bb-cape-overlays/suite/jessie/debian/dtbo_bb-cape-overlays

it'll just exit if the old dtbo script exists..

Regards,

malc0, 2017-04-26:
Thanks, that works a treat.

closed 2017-04-26 by malc0