RobertCNelson/bb.org-overlays #15: fail to inizialize DLPDLCR2000 EVM cape

kaputo, 2017-08-25:
in some board the output pin "P9.15" connected to PROJ_ON_EXT at startup is low, this prevents the correct initialization of  DLPC2607 chip.

RobertCNelson, 2017-08-25:
@kaputo please provide the output of version.sh:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

kaputo, 2017-08-26:
root@beaglebone:/# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[8e0b1e6f135ff3c30468f49a7a952c98be97219a]
eeprom:[A335BNLTBWA51650BBWG4923]
dogtag:[BeagleBoard.org Debian Image 2017-08-21]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-rc2-00002-g84a7f2]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.07-00002-g61c3ee0fb5]
kernel:[4.9.44-ti-r55]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170821-0rcnee1~stretch+20170821]
pkg:[firmware-ti-connectivity]:[20161130-3]

kaputo, 2017-08-26:
https://e2e.ti.com/support/dlp__mems_micro-electro-mechanical_systems/video_and_data_display/f/947/t/618632