RobertCNelson/bb.org-overlays #18: 4DCAPE43 LCD Cape overlay custom

FabioGenesy, 2017-10-03:
Hi Robert,
I bought a 4DCAPE43 LCD Cape but I've some doubt regarding overlay.
I need to modify the BB-BONE-LCD4-01 that is automatically recognized when LCD CAPE43 is attached and disabled GPIO0_15 (Enter Button).
I then modify the corresponding dbto in /lib/firmware (decompiled and recompiled) but the button still works. I also removed the dbto in /lib/firmware folder but it's loaded the same at startup.
From where this file (BB-BONE-LCD4-01-00A1.dtbo) is taken? Is it loaded from EEPROM of LCD CAPE 43? How I can apply my custom overlay for this LCD?
Thank you very much!

RobertCNelson, 2017-10-03:
@FabioGenesy

*.dtbo objects are stored in 2 (and a 3rd legacy location)

1: /lib/firmware/*
2: /boot/initrd.img-`uname -r`
3: /boot/vmlinuz-`uname -r` (legacy)

1: easy to update

2:
```
sudo update-initramfs -uk `uname -r`
```

3: 3.8.13: full kernel rebuild (*.dtbo's are built-in)
```
https://github.com/RobertCNelson/yakbuild
```

Regards,

FabioGenesy, 2017-10-03:
Thank you for the quick answer!
So in my case he is loading it from the kernel. 
When I modify an overlay, I have to update it in all 2(3) location?
What does the second command do (update-initramfs )? Does update the initrd.img file from /lib/firmware? 
How can I disable automatic upload only for the LCD in case?

Thank you so much!

RobertCNelson, 2017-10-03:
For 3.8.13 just update the kernel copy, the other two will be ignored..

update-initramfs scan's /lib/firmware/ to generate initrd.img

Regards,

FabioGenesy, 2017-10-03:
If I understand it well I need to recompile the kernel anyhow.
My kernel is 4.4.54-ti-r93, can I use your link?
I've another doubt. I do command 2 and I update initramfs but now it apply for default the univ-all overlay (not before). Why? This overlay has a conflict with uarts and is a problem for me!

RobertCNelson, 2017-10-03:
Okay, with v4.4.x just ignore #3.

univ-all only get's loaded last in the boot cycle, after nothing else has loaded.. So if you load your lcd, univ-all won't get loaded..

Ps, with univ-all, you can enable the uart by:

config-pin P9_11 uart
config-pin P9_13 uart

and /dev/ttyS4 will come alive..

Regards,

FabioGenesy, 2017-10-20:
Hi Robert thank you so much, 
I solved several problems and understood better how the device tree works!
I notice however a strange behavior with this display. 
If I read an adc channel with the /iio:device0/in_voltage_n_raw descriptor 
the mouse pointer on the screen moves randomly.
I know that touchscreen input are adc channels but the monitor use only channels [0-3].
If I look the BB-BONE-LCD4-01-00A1.dtbo decompiled for me is correct.
Here is the extract:
```
fragment@4 {
		target = <0xdeadbeef>;

		__overlay__ {
			status = "okay";

			tsc {
				ti,wires = <0x4>;
				ti,x-plate-resistance = <0xc8>;
				ti,coordinate-readouts = <0x5>;
				ti,wire-config = <0x0 0x11 0x22 0x33>;
				ti,charge-delay = <0x400>;
			};

			adc {
				ti,adc-channels = <0x4 0x5 0x6 0x7>;
				ti,chan-step-opendelay = <0x98 0x3ffff 0x98 0x0>;
				ti,chan-step-sampledelay = <0xff 0x0 0xf 0x0>;
				ti,chan-step-avg = <0x10 0x2 0x4 0x8>;
			};
		};
	};

```
My only doubt is that ADC channels on beaglebone external pinout are 7 but on device description there is a channel 0x7 (I think that first channel is 0 and for me last channel is 6)

Thank you in advance