RobertCNelson/bb.org-overlays #19: Problems with enabling customized overlay and onboard WIFI on BBBW, and auto-loading overlay.

zongyaojin, 2017-10-04:
Hello Robert,

I wrote [a `.dts` file](https://github.com/zongyaojin/research-bb_overlays/blob/v1/xlearn_v1.dts) to enable some functions (8 PWMs, 2 eQEPs, 2 I2Cs, 7 AINs) on a Beaglebone Black. And here is my [`uEnv.txt` file](https://github.com/zongyaojin/research-bbb/blob/master/uEnv/current/uEnv.txt) which helps me to disable the HDMI to avoid conflicts. 

It worked on my BBB without any problems. However, I have a few questions, and I was hoping that you could give me some guidance. My kernel version is: `Linux arm 4.4.68-ti-r108 #1 SMP Fri Jun 9 18:52:39 UTC 2017 armv7l armv7l armv7l GNU/Linux`

1. How to change my `uEnv.txt` accordingly on a Beaglebone Black **Wireless**, such that I'm able to activate all these functions and enable the onboard WIFI. Currently this combo of `.dts` and `uEnv.txt` worked fine on Beaglebone Black Wireless, except that the onboard WIFI doesn't get activated. So, what should I do to fix that?

2. How to get the .dts file I wrote automatically loaded on system boot? Currently I'm using [a shell script](https://github.com/zongyaojin/research-bb_overlays/blob/v1/deploy_xlearn_v1) to activate the overlay. I've tried the methods you mentioned on the wiki page, but I wasn't able to get it loaded on boot.

3. It seems that every time I activate my overlay, the `pwmchip#` directories in `/sys/devices/platform/ocp/4830#000.epwmss/4830#200.pwm/pwm/` paths change their number randomly. For example, one time a PWM directory can be `/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip0`, another time the directory can be `/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip2`, 0 changes to 2. Why that happens? Is there a way to fix those numbers?

Thank you very much for your time and help!

Sincerely,

Zongyao

RobertCNelson, 2017-10-04:
1: first of all, use "U-Boot" overlays, what you are trying to do on the "Wireless" became and O^X problem, as every "board" needed 6 varitations..  So yeah, what you are trying to do with "kernel" overlays is NOT supported.

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

2: Also solved with U-Boot overlays:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Override_external_capes

3: yeah... Documented here:
https://github.com/beagleboard/bb.org-overlays/blob/master/examples/cape-unversal-pwm.txt

Essentially read this value "first"
```
#P9.21/P9.22
#ls -lh /sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/
#P9.14/P9.16
#ls -lh /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/
#P8.13/P8.16
#ls -lh /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/
```
It'll then point you to the correct "pwmchipX"..

Regards,

RobertCNelson, 2017-10-04:
For reference this is the basic U-Boot Overlays section we ship in /boot/uEnv.txt by default

```
###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###
```

zongyaojin, 2017-10-04:
Thank you very much for your quick reply! All problems solved like magic! 

Just a bit documentation in hopes that it would be helpful for people running into the same problems, who, like me, doesn't understand kernel and universal bootloader very well and may take some time to figure out the steps based on your guidance. 

What I did was (I'm using Ubuntu 16.04):

1. Updated my `/boot/uEnv.txt` with (since I have all the functions I needed in my overlay file, and some of them conflict with HDMI and some default stuff):

> enable_uboot_overlays=1
> dtb_overlay=/lib/firmware/<custom_overlay_name>.dtbo
> disable_uboot_overlay_emmc=1
> disable_uboot_overlay_video=1
> disable_uboot_overlay_audio=1
> disable_uboot_overlay_adc=1
> disable_uboot_overlay_addr0=1
> disable_uboot_overlay_addr1=1
> disable_uboot_overlay_addr2=1
> disable_uboot_overlay_addr3=1

2. Updated my U-Boot with:

> (Kernel Overlays are going bye-bye, too many bugs, too many race conditions, no kernel maintainers interested. [https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays) )

```
ubuntu@arm:~$ cd /opt/scripts/tools/
ubuntu@arm:/opt/scripts/tools$ git pull
ubuntu@arm:/opt/scripts/tools$ sudo ./version.sh
ubuntu@arm:~$ cd /opt/scripts/tools/developers/
ubuntu@arm:/opt/scripts/tools/developers$ sudo ./update_bootloader.sh
ubuntu@arm:/opt/scripts/tools/developers$ sudo reboot
```

3. And a last side note, since the new U-Boot aims at replacing kernel overlays, the following "traditional" overlay loading won't work:

```
ubuntu@arm:/lib/firmware$ export SLOTS=/sys/devices/platform/bone_capemgr/slots
ubuntu@arm:/lib/firmware$ echo compiled_custom_overlay_name > $SLOTS
```

Again, thank you very much! My gratitude is really beyond description!

Very respectfully,

Zongyao

closed 2017-10-04 by zongyaojin