RobertCNelson/bb.org-overlays #20: HDMI audio osc disabled with standard overlay on BBBW

acourdavault, 2017-11-22:
Hello @RobertCNelson 

I think i may have an issue with the overlays loaded at the startup of my BeagleBone Black
- platform: BBBW
- OS: Debian Stretch from official image of the beaglebone site
- OS version: `Linux beaglebone 4.9.45-ti-r57 #1 SMP PREEMPT Fri Aug 25 22:58:38 UTC 2017 armv7l`

My issue is that HDMI works but not Audio (which i want to use with an external i2s DAC).
I have an oscilloscope to debug the i2s protocol.
I managed to get it to work (audio to i2s) a month ago but i cannot reproduce it. 

What makes me think it s related to the overlays is that the Audio Oscillator Enable pin (gpio1.27) is lo.
I check this with this tool: https://github.com/mvduin/bbb-pin-utils
```
audio osc enable                  27 fast    down 7 gpio 1.27 lo >>  clk_mcasp0 (mcasp0_pins)
P9.31 / hdmi audio clk           100 fast    down 0 asp 0 tx clk     mcasp@48038000 (mcasp0_pins)
P9.29 / hdmi audio fs            101 fast     up  0 asp 0 tx fs      mcasp@48038000 (mcasp0_pins)
P9.28 / hdmi audio data          103 fast    down 2 asp 0 data 2     mcasp@48038000 (mcasp0_pins)
P9.25 / audio osc                107 fast rx  up  0 asp 0 tx hclk    mcasp@48038000 (mcasp0_pins)
```

I read your comments on this issue: https://github.com/adafruit/adafruit-beaglebone-io-python/issues/127

As suggested in several places i updated my bootloader and the kernel.

Also following this note https://github.com/beagleboard/bb.org-overlays/issues/43
I tried to move to the "non TI" version of the OS.
This did not change anything either.

`./version.sh`
```
git:/opt/scripts/:[2ce750d881941c5189db9e189af90517e11c079f]
eeprom:[A335BNLTBWA51645BBWG0838]
dogtag:[BeagleBoard.org Debian Image 2017-08-31]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.11-00002-g7b415acfc1]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.11-rc3-00002-g73df7f]
kernel:[4.9.63-bone8]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

Command Line arguments:
I have also tried in `uEnv.txt` to add `hdmi.audio=1` to the command line, this did not change anything.

As this may be important the screen resolution is (`xrandr --verbose`)
`1024x768 (0x1b6)  0.000MHz *current`

And looked at the overlays, 
The overlays loaded are here (`dmesg | grep bone`):
```
[    3.925752] usb usb1: Manufacturer: Linux 4.9.63-bone8 musb-hcd
[    4.449171] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,BWA5,1645BBWG0838'
[    4.449188] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    4.473202] bone_capemgr bone_capemgr: slot #0: No cape found
[    4.501906] bone_capemgr bone_capemgr: slot #1: No cape found
[    4.532245] bone_capemgr bone_capemgr: slot #2: No cape found
[    4.562580] bone_capemgr bone_capemgr: slot #3: No cape found
[    4.568398] bone_capemgr bone_capemgr: initialized OK.
[    8.581307] systemd[1]: Set hostname to <beaglebone>.
```

**I think the issue is that GPIO 1. 27 is LO which desactivated the audio oscillator**
The question is **how to turn it on**

Why would the Overlays not activate HDMI audio.

Ive also looke at the `cape-universal cape-universala cape-universaln` but without success.

Do you have any suggestion of an overlay load error on this?

Thank you

acourdavault, 2017-11-22:
I wonder if it may be important to add something to turn on GPIO1.27, as I've check it s lo on the hardware

I've seen it in 
https://github.com/beagleboard/devicetree-source/blob/master/arch/arm/boot/dts/cape-boneblack-hdmi-00A0.dts
it s a PULLPUP but it s definitly not at 3.3V on my board

any advice?

RobertCNelson, 2017-11-22:
Based on the dts, it should be low:

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/arch/arm/boot/dts/am335x-boneblack.dts?h=v4.9.64#n75

Sorry, i don't have a hdmi-audio based device at work to test this.

Regards,

acourdavault, 2017-11-22:
@RobertCNelson 
I see, thank you for your answer,

Should I try to modify this overlay to set it?

When you say it 's low by default, is it because of this line:
```
		enable-gpios = <&gpio1 27 0>; /* BeagleBone Black Clk enable on GPIO1_27 */
```
that i should change to `<&gpio1 27 1>;`

Or would there be a non-intrusive way to set this value?
*i would definitely prefer another way*

Thank you again

acourdavault, 2017-11-23:
Hello @RobertCNelson 

Thank you I found the issue the overlays are working fine,
It's not possible to force this pin status,

However with the proper resolution CEA for my screen it gets to hi when sound is playing only and not at startup.

Sorry for the noise

closed 2017-11-23 by acourdavault

jjghjjgh, 2018-03-28:
Hello @acourdavault

I just saw in your previous post that you were able to connect audio DAC on BBB. I also want to enable audio DAC with i2s interface on BBB running the same kernel 4.9 as yours. I was trying to load BB-BONE-AUDI-02 overlay which is based on TLV320AIC3106 codec. May i know u also used tha same.If yes i had some doubts related to loading the same audio audio cape on new kernel 4.9. Please respond. waiting for your response.
Thanks,
Jithu