RobertCNelson/bb.org-overlays #30: dtc-overlay.sh is checking outdated git-core package

QuickJack, 2019-06-18:
This happens at https://github.com/RobertCNelson/bb.org-overlays/blob/38609b9006cf2d6d74b0e2857e3a11be46376f41/dtc-overlay.sh#L14

apt-get install reports **Note, selecting 'git' instead of 'git-core'**. This issue creates problems when running the script as non root in a proxied environment. The http_proxy environment variable is not available in this case.

RobertCNelson, 2019-06-18:
@QuickJack , cool thanks, today most distro's should be at atleast 1.4.4 b default, so that script isn't really needed. ;)

closed 2019-06-18 by RobertCNelson

QuickJack, 2019-06-26:
I am using the latest version of the gitlab/gitlab-runner docker image. It is based on Ubuntu 16.04. Package device-tree-compiler has version 1.4.0.

As a workaround I have added the following line to my Dockerfile:

`RUN echo 'Defaults env_keep += "http_proxy https_proxy"' >> /etc/sudoers`

This will prevent apt-get from trying to install the git-core dummy package using sudo which doesn't work in a proxied environment.