RobertCNelson/bb.org-overlays #32: I get the FDT_ERR_NOTFOUND error when trying to use a custom uboot overlay

asimonf, 2019-08-22:
I've been using a custom cape designed by a user with debian 8 and the 3.8 kernel branch just fine for the last year. Recently, I decided to try to use in newer debian releases because I need libraries and utils available since scratch. I understood that the newer kernels deprecated kernel overlays in favor of uboot overlays, but I haven't been able to use this overlay with uboot without getting the error mentioned on the title.

Here's the overlay code:

```
/dts-v1/;
/plugin/;

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black", "ti,beaglebone-green";
	part-number = "pru_arvid";
	version = "00A0";

	/* Define pins and resources this config sets up. */
	/* pin numbers on header P8 or P9 */

	exclusive-use = 
	"P8.45", /* GPIO2_6 */
	"P8.46", /* GPIO2_7 */
	"P8.43", /* GPIO2_8 */
	"P8.44", /* GPIO2_9 */
	"P8.41", /* GPIO2_10 */

	"P8.42", /* GPIO2_11 */
	"P8.39", /* GPIO2_12 */
	"p8.40", /* GPIO2_13 */
	"p8.37", /* GPIO2_14 */
	"p8.38", /* GPIO2_15 */

	"p8.18", /* GPIO2_1 */
	"p8.9",  /* GPIO2_5 */
	"p8.10", /* GPIO2_4 */
	"p8.7",  /* GPIO2_2 */
	"p8.8",  /* GPIO2_3 */

	"p8.28", /* GPIO2_24 - sync */

	"p9.27", /* GPIO3_19 TATE switch */
	"p9.28", /* GPIO3_17 Coin button */
	"p9.25", /* GPIO3_21 Start button */

	"pru1" ; /* PRU unit 1 */

	

	fragment@0 {
		target = <&am33xx_pinmux>;

		__overlay__ {
			status = "okay";

			arvid_pru_pins: pinmux_arvid_pru_pins {
				/* PRU Low latency IO */
				pinctrl-single,pins = <
					0x0A0 0x07  /* P8.45 mode 7 -> output */
					0x0A4 0x07  /* P8.46 mode 7 -> output */
					0x0A8 0x07  /* P8.43 mode 7 -> output */
					0x0AC 0x07  /* P8.44 mode 7 -> output */
					0x0B0 0x07  /* P8.41 mode 7 -> output */

					0x0B4 0x07  /* P8.42 mode 7 -> output */
					0x0B8 0x07  /* P8.39 mode 7 -> output */
					0x0BC 0x07  /* P8.40 mode 7 -> output */
					0x0C0 0x07  /* P8.37 mode 7 -> output */
					0x0C4 0x07  /* P8.38 mode 7 -> output */

					0x08C 0x07  /* P8.18 mode 7 -> output */
					0x09C 0x07  /* P8.9  mode 7 -> output */
					0x098 0x07  /* P8.10 mode 7 -> output */
					0x090 0x07  /* P8.7  mode 7 -> output */
					0x094 0x07  /* P8.8  mode 7 -> output */

					0x0E8 0x07  /* P8.28 mode 7 -> output */

					0x1A4 0x37  /* P9.27 mode 37 -> input, pullup */
					0x19C 0x37  /* P9.28 mode 37 -> input, pullup */
					0x1AC 0x37  /* P9.25 mode 37 -> input, pullup */
				>;
			};

		};
	};

	fragment@1 {
		target = <&pruss>;
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&arvid_pru_pins>;
			enabled = <1>;
			/* 
			NOTE: all gpioX controllers are numbered from 1 !
			That means GPIO2 is actually specified here as 'gpio3' !
			*/

			hv_sync {
				pin-names = "Arvid:hv_sync";
				gpios = <&gpio3 24 0>;
			};

			buttons {
				pin-names = "Arvid:tate_switch", "Arvid:coin_button", "Arvid:start_button";
				gpios = <&gpio4 27 0    &gpio4 28 0     &gpio4 25 0>;
			};

			red_bits {
				pin-names = "Arvid:red_0", "Arvid:red_1", "Arvid:red_2", "Arvid:red_3", "Arvid:red_4";
				gpios     = <&gpio3 11 0    &gpio3 12 0    &gpio3 13 0   &gpio3 14 0   &gpio3 15 0>;
			};

			green_bits {
				pin-names = "Arvid:green_0", "Arvid:green_1", "Arvid:green_2", "Arvid:green_3", "Arvid:green_4";
				gpios     = <&gpio3 6 0       &gpio3 7 0       &gpio3 8 0       &gpio3 9 0       &gpio3 10 0>;
			};

			blue_bits {
				pin-names = "Arvid:blue_0", "Arvid:blue_1", "Arvid:blue_2", "Arvid:blue_3", "Arvid:blue_4";
				gpios     = <&gpio3 1 0      &gpio3 2 0      &gpio3 3 0      &gpio3 4 0      &gpio3 5 0>;
			};

		};
	};

};
```

Here's my version output:

```
git:/opt/scripts/:[5b2e16aa1e5c0f627f1d48a6dd1c13b446b9f53b]
eeprom:[A335BNLTSE0A2618BBE70DE5]
model:[SanCloud_BeagleBone_Enhanced]:WiFi AP Enabled:[https://github.com/lwfinger/rtl8723bu]
dogtag:[BeagleBoard.org Debian Image 2019-08-11]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
kernel:[4.19.59-ti-r25]
nodejs:[v10.15.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190812.0-0rcnee0~buster+20190812]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~buster+20190227]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190114-1]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.705873] remoteproc remoteproc0: wkup_m3 is available
[    1.870955] remoteproc remoteproc0: powering up wkup_m3
[    1.871084] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.874814] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.162939] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.176627] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 003: ID 0bda:b720 Realtek Semiconductor Corp.
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```
and here's the output I get when the boot fails:

```
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: uboot loading of [/lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_emmc=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_video=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_adc=1]...
uboot_overlays: loading /lib/firmware/AM335X-PRU-UIO-00A0.dtbo ...
1013 bytes read in 158 ms (5.9 KiB/s)
uboot_overlays: [dtb_overlay=/lib/firmware/pru_arvid-00A0.dtbo] ...
uboot_overlays: loading /lib/firmware/pru_arvid-00A0.dtbo ...
2710 bytes read in 2404 ms (1000 Bytes/s)
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
uboot_overlays: cape universal disabled, external cape enabled or detected...
loading /boot/initrd.img-4.19.59-ti-r25 ...
7757010 bytes read in 545 ms (13.6 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet] ...
debug: [bootz 0x82000000 0x88080000:765cd2 88000000] ...
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
** Invalid partition 2 **
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
switch to partitions #0, OK
```

The firmware blob was built by pulling the this repo, copying the source to the `src/arm` folder and running make with the name file, then copied to the firmwares directory. I've been trying for several days now searching the web for answers, but I haven't found any hints as to why this could be failing. The error code is not particularly descriptive.

Any help would be appreciated. Cheers!

RobertCNelson, 2019-08-22:
```			/* 
			NOTE: all gpioX controllers are numbered from 1 !
			That means GPIO2 is actually specified here as 'gpio3' !
			*/
```

That's a 3.8 only thing.. fix that...

asimonf, 2019-08-23:
Thanks! Now I managed to boot, but now uio is not working. The following appears on dmesg:

```
[   31.063908] pinctrl-single 44e10800.pinmux: pin PIN40 already requested by ocp:P8_45_pinmux; cannot claim for 4a300000.pruss
[   31.254998] pinctrl-single 44e10800.pinmux: pin-40 (4a300000.pruss) status -22
[   31.364872] pinctrl-single 44e10800.pinmux: could not request pin 40 (PIN40) from group pinmux_arvid_pru_pins  on device pinctrl-single
[   31.522015] pruss_uio 4a300000.pruss: Error applying setting, reverse things back
```

I do assign pin 40 on my overlay, but I disabled all other overlays in uEnv.txt and the uio overlay doesn't seem to request pin 40 so I don't know what gives... Any help would be appreciated.

asimonf, 2019-08-24:
As an update, that set of pins is claimed by pin group `nxp_hdmi_bonelt_pins` but I'm disabling hdmi on uEnv.txt. Here's my uEnv.txt file:

```
uname_r=4.19.59-bone36

enable_uboot_overlays=1
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo

dtb_overlay=/lib/firmware/pru_arvid-00A0.dtbo

cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 cape_universal=false quiet
```

My boot log shows this:

```
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=] ...
debug: [uboot_base_dtb_univ=am335x-sancloud-bbe-uboot.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-sancloud-bbe-uboot.dtb] ...
loading /boot/dtbs/4.19.59-bone36/am335x-sancloud-bbe.dtb ...
53230 bytes read in 161 ms (322.3 KiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: uboot loading of [/lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_emmc=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_video=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_adc=1]...
uboot_overlays: loading /lib/firmware/AM335X-PRU-UIO-00A0.dtbo ...
1035 bytes read in 1417 ms (0 Bytes/s)
uboot_overlays: [dtb_overlay=/lib/firmware/pru_arvid-00A0.dtbo] ...
uboot_overlays: loading /lib/firmware/pru_arvid-00A0.dtbo ...
2732 bytes read in 321 ms (7.8 KiB/s)
uboot_overlays: add [enable_uboot_cape_universal=1] to /boot/uEnv.txt to enable...
```

I've tried with both and without universal cape support enabled and it makes no difference. I tried with an older 4.14 kernel and this newest 4.19 branch and both the mainline and the ti versions of the latest one and the error persists. I can't load a cape that uses pins from the hdmi group even if the overlay is disabled.

Any help, please?

asimonf, 2019-08-24:
I tried the same μSD card on a Beaglebone Black and it worked perfectly. Same on a Beaglebon Green. The issue is then on the Sancloud Enhanced Beaglebone. Any ideas where the problem might be? I don't mind experimenting on the overlays to see if I can get to the bottom of it. 

I could really use the extra ethernet bandwidth if it can be helped.

asimonf, 2019-10-13:
Took me a while to get around to test this. It didn't work on the SanClound BB. I added this:

```
       /*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@1 {
		target = <&ocp>;
		__overlay__ {
			P8_45_pinmux { status = "disabled"; };	/* lcd_data0 */
			P8_46_pinmux { status = "disabled"; };	/* lcd_data1 */
			P8_43_pinmux { status = "disabled"; };	/* lcd_data2 */
			P8_44_pinmux { status = "disabled"; };	/* lcd_data3 */
			P8_41_pinmux { status = "disabled"; };	/* lcd_data4 */
			P8_42_pinmux { status = "disabled"; };	/* lcd_data5 */
			P8_39_pinmux { status = "disabled"; };	/* lcd_data6 */
			P8_40_pinmux { status = "disabled"; };	/* lcd_data7 */
			P8_37_pinmux { status = "disabled"; };	/* lcd_data8 */
			P8_38_pinmux { status = "disabled"; };	/* lcd_data9 */
			P8_36_pinmux { status = "disabled"; };	/* lcd_data10 */
			P8_34_pinmux { status = "disabled"; };	/* lcd_data11 */
			P8_35_pinmux { status = "disabled"; };	/* lcd_data12 */
			P8_33_pinmux { status = "disabled"; };	/* lcd_data13 */
			P8_31_pinmux { status = "disabled"; };	/* lcd_data14 */
			P8_32_pinmux { status = "disabled"; };	/* lcd_data15 */
			P8_27_pinmux { status = "disabled"; };	/* lcd_vsync */
			P8_29_pinmux { status = "disabled"; };	/* lcd_hsync */
			P8_28_pinmux { status = "disabled"; };	/* lcd_pclk */
			P8_30_pinmux { status = "disabled"; };	/* lcd_ac_bias_en */
		};
	};
```

I also changed the previous `fragment@1` to `fragment@2`. As soon as the device booted I got this error: 

```
[   59.727879] pinctrl-single 44e10800.pinmux: pin PIN40 already requested by 0-0070; cannot claim for 4a300000.pruss
[   60.007360] pinctrl-single 44e10800.pinmux: pin-40 (4a300000.pruss) status -22
[   60.165994] pinctrl-single 44e10800.pinmux: could not request pin 40 (PIN40) from group pinmux_arvid_pru_pins  on device pinctrl-single
```

Executing `cat /sys/kernel/debug/pinctrl/44e10800.pinmux-pinctrl-single/pingroups` I got this (among other groups):

```
group: nxp_hdmi_bonelt_pins
pin 40 (PIN40)
pin 41 (PIN41)
pin 42 (PIN42)
pin 43 (PIN43)
pin 44 (PIN44)
pin 45 (PIN45)
pin 46 (PIN46)
pin 47 (PIN47)
pin 48 (PIN48)
pin 49 (PIN49)
pin 50 (PIN50)
pin 51 (PIN51)
pin 52 (PIN52)
pin 53 (PIN53)
pin 54 (PIN54)
pin 55 (PIN55)
pin 56 (PIN56)
pin 57 (PIN57)
pin 58 (PIN58)
pin 59 (PIN59)

group: pinmux_arvid_pru_pins
pin 40 (PIN40)
pin 41 (PIN41)
pin 42 (PIN42)
pin 43 (PIN43)
pin 44 (PIN44)
pin 45 (PIN45)
pin 46 (PIN46)
pin 47 (PIN47)
pin 48 (PIN48)
pin 49 (PIN49)
pin 35 (PIN35)
pin 39 (PIN39)
pin 38 (PIN38)
pin 36 (PIN36)
pin 37 (PIN37)
pin 58 (PIN58)
pin 105 (PIN105)
pin 103 (PIN103)
pin 107 (PIN107)
```

Please bare in mind that I've disabled video in uEnv.txt. Any help is appreciated.

asimonf, 2019-10-16:
I figured it out. Thanks for your help!

closed 2019-10-16 by asimonf