RobertCNelson/bb.org-overlays #33: RE: Ideas and Overlay Stuff

silver2row, 2019-09-06:
Hello,

Is there a u-boot overlay for https://www.seeedstudio.com/5-Inch-BeagleBone-Green-LCD-Cape-with-Resistive-Touch-p-2642.html?

I am just asking. I do not expect anyone to make one just for one person. If there is, awesome! If not, I guess I can try to make something happen.

Seth

silver2row, 2019-09-06:
Hello Once More,

I have not found any documents on this specific Cape. Their, the Seeed Studio people, document section is lacking.

Seth

RobertCNelson, 2019-09-06:
@silver2row Seeed first added support for this back on April 12, 2016:

https://github.com/RobertCNelson/bb.org-overlays/commits/master/src/arm/BB-BONE-LCD5-01-00A1.dts

REgards,

closed 2019-09-06 by RobertCNelson

silver2row, 2019-09-06:
Hello Sir,

Thank you.

Seth