RobertCNelson/bb.org-overlays #34: RS485 Overlay for Waveschare CAN/RS485 Cape

Bouni, 2019-09-16:
I try to get RS485 with the [Waveshare CAN/RS485](https://www.waveshare.com/wiki/RS485_CAN_CAPE) to work on the latest Debian Buster Image (Debian 10.0 2019-07-07 4GB SD IoT) but had no success so far.

Since 2016 I had this cape running with a 3.8.XX Kernel but thought its time to update.
Unfortunately a lot changed since than, so my approach from back then does no longer work.

On 3.8 I just enabled UART4 using `cape_enable=capemgr.enable_partno=BB-UART4` in the uEnv.txt and then used this ioctl (python) to switch RX/TX:

```
TIOCSRS485 = 0x542F
RS485_FLAGS = 0x21 # enable RS485 and use GPIO for RE/DE control
RS485_GPIO_PIN = 7
serial_rs485 = struct.pack('IIIIIIII',
                           RS485_FLAGS,        # config flags
                           0,                  # delay in us before send
                           0,                  # delay in us after send
                           RS485_GPIO_PIN,     # the pin number used for DE/RE
                           0, 0, 0, 0          # padding - space for more values
                           )
fd=s.fileno()
fcntl.ioctl(fd, TIOCSRS485, serial_rs485)
```
The Waveshare Cape uses Pin P9.42 (GPIO0_7) for switching RX/TX.
For I don't know what reason, this no longer works.

I tried to use the [BB-UART4-RS485-00A0.dts](https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART4-RS485-00A0.dts) uBoot overlay without success, but that's no surprise because it specifies P9.27 to switch RX/TX.

So I made a copy of that file and changes all `P9_27` for `P9_42` and `gpio3_19` for `gpio0_7`, compiled it successful, copied it to /lib/firmware and put it into uEnv.txt.

Still no success :-(

By the way, receiving works but again, that's no surprise.

Any ideas what i do wrong?

RobertCNelson, 2019-09-16:
@Bouni the default serial_8250_omap does a bad job at real RS485, sadly you'll need to rebuild the kernel and switch to CONFIG_SERIAL_OMAP.

Disable:
```
CONFIG_SERIAL_8250_OMAP=y
CONFIG_SERIAL_8250_OMAP_TTYO_FIXUP=y
```

Enable:
```
CONFIG_SERIAL_OMAP=y
```
Regards,

Bouni, 2019-09-17:
@RobertCNelson 
Thanks for the fast answer! 
Are there instructions somewhere on how to do that? Never built a kernel before.
I assume that its smart to do it on a PC with more power an cross compile, right?

ffeldbauer, 2020-10-30:
Hey,

I'm using the Debian 10 image with Kernel version 5.4.66-ti-r18.
Based on the [UART4-RS485 overlay](https://github.com/RobertCNelson/bb.org-overlays/blob/master/src/arm/BB-UART4-RS485-00A0.dts), I wrote an overlay for UART5 using `P8.37` (Tx), `P8.38` (Rx), `P8.32` (de/re).

I have the problem that `gpio0_11` doesn't go high when transmitting.
Is this still a problem of `serial_8250_omap` in Kernel version 5.4?

Regards,
Florian