RobertCNelson/bb.org-overlays #36: Seeing: "Missing interrupt-parent" compiling dtbs

buzmeg, 2020-05-19:
When I do a "make all" on Debian 10, I see messages like these when compiling:

`AM335X-PRU-RPROC-4-19-TI-PRUCAPE-00A0.dtbo: Warning (interrupts_property): /fragment@3/__overlay__/ocp/pruss-soc-bus@4a326004/pruss@4a300000: Missing interrupt-parent`

I also sometimes see:

`AM335X-PRU-RPROC-4-19-TI-PRUCAPE-00A0.dtbo: Warning (chosen_node_is_root): /fragment@0/__overlay__/chosen: chosen node must be at root node
`

I don't know the dtb stuff well enough to know if that is important or not.

Thanks.

RobertCNelson, 2020-05-19:
@buzmeg you should not actually see the 2nd one anymore... 

https://github.com/beagleboard/bb.org-overlays/blob/master/Makefile#L61

The first, one well that's the way TI defined the node in their v4.19.x-ti branch, they should have had an interrupt-parent in it according to the spec, but it doesn't matter as there driver knows what todo..

Regards,

buzmeg, 2020-05-19:
Thanks.  Much appreciated.

closed 2020-05-19 by buzmeg