RobertCNelson/bb.org-overlays #37: BBBMini dtb not working on kernel 4.14 and 4.19

juvinski, 2020-10-16:
**Describe the bug**
Hi, using 4.14 or 4.19 versions, the bbbmini - BB-BBBMINI-00A0 or the am335x-boneblack-bbbmini.dtb is not working.
I have this (am335x-boneblack-bbbmini.dtb) with kernel 4.4 working prety fine but when I'm trying to use the 4.1X no.

Image:  Debian Stretch Console Image 2020-09-21

Beaglebone Black and Green

BBBMini cape

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[20c080bc91f243ef852a704758788a31083b0f9e]
eeprom:[A335BNLTBBG1BBG219100437]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Stretch Console Image 2020-09-21]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBBMINI-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.14.198-bone-rt-r40]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-BBBMINI-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~stretch+20200814]
pkg:[bb-customizations]:[1.20200826.0-0~stretch+20200826]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~stretch+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~stretch+20200813]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio admin spi iio i2c docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.806607] remoteproc remoteproc0: wkup_m3 is available
[    0.867878] remoteproc remoteproc0: powering up wkup_m3
[    0.867899] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    0.871038] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.578594] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.588766] gpio-of-helper ocp:cape-universal: ready
END

root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[20c080bc91f243ef852a704758788a31083b0f9e]
eeprom:[A335BNLTBBG1BBG219100437]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Stretch Console Image 2020-09-21]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.14.198-bone-rt-r40]
device-tree-override:[dtb=am335x-boneblack-bbbmini.dtb]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~stretch+20200814]
pkg:[bb-customizations]:[1.20200826.0-0~stretch+20200826]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~stretch+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~stretch+20200813]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio admin spi iio i2c docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.864836] remoteproc remoteproc0: wkup_m3 is available
[    0.927481] remoteproc remoteproc0: powering up wkup_m3
[    0.927502] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    0.930819] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.582686] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.592719] gpio-of-helper ocp:cape-universal: ready
END

juvinski, 2020-10-16:
Another issue - I create a config to use pins instead the dtb file, with that, I could you ardupilot software - but I'm having a extreme high load average when using any I2C device connected, and the same ardupilot version (4.06) on the 4.4 kernel and the am335x-boneblack-bbbmini.dtb.

Thanks for your help

juvinski, 2020-12-17:
HI @RobertCNelson ,

any comment or test or something I can check to see what this dtb is not working ?

Thanks a lot

juvinski, 2020-12-22:
Hi.

after several tests, on last 4.14.209 the BBBMini DTB is working fine, but in 4.19 no.
There are any clue ? test or something I can do ?

juvinski, 2020-12-22:
Hi @RobertCNelson 

I find something. I was testing on a BBG and now I'm testing on BBB, and with the bbbmini on BBB everything is working, but in the BBG not. Any difference on am335x-boneblack-bbbmini.dts for black and green I should considering ?

Thanks for your help.

RobertCNelson, 2020-12-22:
Does the bbbmini use the vcc pins, I forget which at the bbg doesn't have wired..

juvinski, 2020-12-22:
Hi @RobertCNelson ,

I'm powering the bbb's by rail - the cape provides the power. The question is about the dtb is not loading on BBG and working on BBB - I was looking and I found under the **am335x-boneblack-bbbmini.dts** on the compatible flag was missing the bbg, - the line is now - compatible = "ti,am335x-bone-green", "ti,am335x-bone-black", "ti,am335x-bone", "ti,am33xx"
but even with that the dtb file are not correctly loaded.
Next change will be at the model line to insert green instead black.

Thanks.

juvinski, 2020-12-22:
Hi @RobertCNelson,

after digging, I found this:
The main problem here is with some bbgreen boards with more older uboot the lines boot overlays and cape universal doesn't make any diference and the dtb is working with both boards (BBB and BBG).
In some boards with uboot updated I had to comment the boot overlays and cape universal, with that the DTB is loaded correctly and everything is working.

I made another update - update to kernel 4.19 and with same behaviors I describe above - with older uboot the dtb is working but the SPI devices are named correcly - SPIDEV2.0 and 2.0 instead 1.0 and 1.1. 
But I'm facing problem with the PRU - the PRU is not loading.

Any clue ?

Best regards

juvinski, 2020-12-23:
One more detail.
On BBB with 4.19, after several attempts - 45 attempts, the PRU is loading, until this 45 number I'm having this error message:
[   63.551545] pgd = 040f9bba
[   63.551549] [b6f16000] *pgd=97028831, *pte=4a324143, *ppte=4a324832
[   64.995780] Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6f0f000

I will try the same test tomorow on the BBG