RobertCNelson/bb.org-overlays #38: Regression in Kernel 4.19

amilcarlucas, 2020-10-22:
I'm running the **rcn-ee.net Ubuntu Console Image 2020-04-09** and it does not auto-detect my three capes:

- BBORG_COMMS-00A2
- BBORG_PROTO-00A2
- BBORG_RELAY-00A2

I am running an older image in another board and it works fine there:
```
> sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[eed310ea4502de730d275f1a188eda475b181f0c]
eeprom:[A335BNLT000C1941BBBG0488]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2019-06-15]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
UBOOT: Loaded Overlay:[BBORG_PROTO-00A2]
UBOOT: Loaded Overlay:[BBORG_RELAY-00A2]
kernel:[4.14.108-ti-r107]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~bionic+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~bionic+20200702]
pkg:[kmod]:[24-1ubuntu3.4rcnee0~bionic+20200525]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BBORG_RELAY,BBORG_COMMS,BBORG_PROTO, coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.293886] remoteproc remoteproc0: wkup_m3 is available
[    1.502442] remoteproc remoteproc0: powering up wkup_m3
[    1.502558] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.506488] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.048179] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.049426] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 003: ID 0781:5591 SanDisk Corp. 
Bus 001 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

On this Other board (using the same capes) I get:
```
> sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[20c080bc91f243ef852a704758788a31083b0f9e]
eeprom:[A335BNLT000C1940BBBG0782]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net Ubuntu Console Image 2020-04-09]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-r51]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20201021.1-0~bionic+20201021]
pkg:[bb-customizations]:[1.20200826.0-0~bionic+20200826]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~bionic+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~bionic+20200813]
pkg:[kmod]:[24-1ubuntu3.5rcnee0~bionic+20200810]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   72.398833] remoteproc remoteproc0: wkup_m3 is available
[   72.410817] remoteproc remoteproc0: powering up wkup_m3
[   72.410840] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   72.411112] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.825117] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.826850] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 004: ID 046a:00a1 Cherry GmbH 
Bus 001 Device 003: ID ffff:5678  
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

To try and solve this I did:
```
cd /opt/scripts/tools
git pull
./update_kernel.sh
cd /opt/source/bb.org-overlays/
git pull
make
sudo make install
```

But still I get noting when I do:
```
cat /proc/cmdline | sed 's/ /\n/g' | grep uboot_detected_capes= || true
```

Why does it work fine in the older board, and fails in the fully updated board? I even did a `sudo apt update && sudo apt upgrade`

amilcarlucas, 2020-11-09:
It looks like i2c-2 is not working at all :( and that causes the capes to not be detected.
Could this be caused by https://github.com/RobertCNelson/Bootloader-Builder/commit/294bc183dfa64c09ac4a83ba66c1e8dc74b922d9 ?

amilcarlucas, 2020-12-17:
You seam to have added i2c permissions on Nov5 but that has not fixed the problem :(

RobertCNelson, 2020-12-17:
@amilcarlucas sorry for the delay, let's please check soemthing:

https://github.com/RobertCNelson/boot-scripts/commit/c48c2833e3556b5d42a069d85c320f5ef37bd255

Please run:

```
debian@bbg-pwr10-ser18:~$ cd /opt/scripts/tools/
debian@bbg-pwr10-ser18:/opt/scripts/tools$ git pull
```

```
debian@bbg-pwr10-ser18:/opt/scripts/tools$ sudo ./version.sh 
git:/opt/scripts/:[c48c2833e3556b5d42a069d85c320f5ef37bd255]
eeprom:[A335BNLT�BBG115096609]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-00002-gc9b3922522 (Aug 24 2020 - 16:42:18 -0500)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.19.142-bone-rt-r56]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20201119.0-0~buster+20201123]
pkg:[bb-customizations]:[1.20201105.0-0~buster+20201110]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~buster+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100]
dmesg | grep remote
[    2.772246] remoteproc remoteproc0: wkup_m3 is available
[    3.013363] remoteproc remoteproc0: powering up wkup_m3
[    3.057783] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    3.085435] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.779515] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.791034] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
debian@bbg-pwr10-ser18:/opt/scripts/tools$
```

as it'll show the build date of u-boot..

Regards,

amilcarlucas, 2021-01-05:
This is the result:

```
git:/opt/scripts/:[fe926a6a4cf5805041878d7df82b58c00b4f902e]
eeprom:[A335BNLT000C1940BBBG0782]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net Ubuntu Console Image 2020-04-09]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-00002-gc9b3922522 (Aug 24 2020 - 16:42:18 -0500)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-NHDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
UBOOT: Loaded Overlay:[BBORG_PROTO-00A2]
UBOOT: Loaded Overlay:[BBORG_RELAY-00A2]
kernel:[4.19.94-ti-r51]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BBORG_COMMS-00A2.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BBORG_PROTO-00A2.dtbo]
uboot_overlay_options:[uboot_overlay_addr6=/lib/firmware/BBORG_RELAY-00A2.dtbo]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20201119.0-0~bionic+20201123]
pkg:[bb-customizations]:[1.20201105.0-0~bionic+20201110]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~bionic+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~bionic+20200813]
pkg:[kmod]:[24-1ubuntu3.5rcnee0~bionic+20200810]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   87.691852] remoteproc remoteproc0: wkup_m3 is available
[   87.702539] remoteproc remoteproc0: powering up wkup_m3
[   87.702563] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   87.702827] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.828962] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.830738] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID ffff:5678  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```