RobertCNelson/bb.org-overlays #40: Minimum required dtc version

klasyc, 2021-03-17:
Hello community,

I am trying to build the overlays using `dtc 1.4.7` which should be supported yet, but my `dtc` reports syntax error at the following files:
```
src/arm/BBORG_GAMEPUP-00A2.dts
src/arm/BBORG_SERVO-00A2.dts
src/arm/BBAI_BB-BONE-LCD4-01-00A1.dts
```
When I delete the files, the rest of project is built successfully.

Expected behavior: Either increase the minimal supported `dtc` version or fix the files above in order to be compatible with older `dtc`.

Thank you.

RobertCNelson, 2021-03-17:
Hi @klasyc , who's 1.4.7 release are you using? git version builds it..

http://gfnd.rcn-ee.org:8080/view/BeagleBoard/job/beagleboard_overlays/job/master/61/console

https://github.com/RobertCNelson/bb.org-overlays/blob/master/jenkins_build.sh

can you share the build error?

Regards,

klasyc, 2021-03-17:
Thank you for a quick response. Finally it turned out I have more  `dtc` binaries in my system and the build was failing when using version 1.4.4 which is not supported.

closed 2021-03-17 by klasyc